PX4/PX4-Devguide #1018: Creating a Fast RTPS Listener application Documentation Issue

amano001, 2020-04-16:
@TSC21 
I have been following the guide to get PX4 running with Gazebo and ROS2 and I have run into another issue with the documentation. 

The last line to call fastrtpsgen shows sensor_combined.idl but the file that is in the idl folder is called sensor_combined_.idl.
``
cd /path/to/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge
mkdir micrortps_listener
cd micrortps_listener
fastrtpsgen -example x64Linux2.6gcc ../micrortps_client/micrortps_agent/idl/sensor_combined.idl
``

Also once I was able to generate the files by adding the _ to the end of sensor_combined I then run into the issue below

``
g++ -c -Wall -fpic -m64 -O2 -std=c++0x -I.   -c sensor_combined_Subscriber.cxx -o output/x64Linux2.6gcc/sensor_combined_Subscriber.o
sensor_combined_Subscriber.cxx: In member function ‘virtual void sensor_combined_Subscriber::SubListener::onNewDataMessage(eprosima::fastrtps::Subscriber*)’:
sensor_combined_Subscriber.cxx:98:45: error: ‘class sensor_combined_’ has no member named ‘gyro_rad’; did you mean ‘gyro_rad_’?
             std::cout << "gyro_rad: " << st.gyro_rad().at(0);
                                             ^~~~~~~~
                                             gyro_rad_
sensor_combined_Subscriber.cxx:99:37: error: ‘class sensor_combined_’ has no member named ‘gyro_rad’; did you mean ‘gyro_rad_’?
             std::cout << ", " << st.gyro_rad().at(1);
                                     ^~~~~~~~
                                     gyro_rad_
sensor_combined_Subscriber.cxx:100:37: error: ‘class sensor_combined_’ has no member named ‘gyro_rad’; did you mean ‘gyro_rad_’?
             std::cout << ", " << st.gyro_rad().at(2) << std::endl;
                                     ^~~~~~~~
                                     gyro_rad_
sensor_combined_Subscriber.cxx:101:53: error: ‘class sensor_combined_’ has no member named ‘gyro_integral_dt’; did you mean ‘gyro_integral_dt_’?
             std::cout << "gyro_integral_dt: " << st.gyro_integral_dt() << std::endl;
                                                     ^~~~~~~~~~~~~~~~
                                                     gyro_integral_dt_
sensor_combined_Subscriber.cxx:102:69: error: ‘class sensor_combined_’ has no member named ‘accelerometer_timestamp_relative’; did you mean ‘accelerometer_timestamp_relative_’?
             std::cout << "accelerometer_timestamp_relative: " << st.accelerometer_timestamp_relative() << std::endl;
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                     accelerometer_timestamp_relative_
sensor_combined_Subscriber.cxx:103:55: error: ‘class sensor_combined_’ has no member named ‘accelerometer_m_s2’; did you mean ‘accelerometer_m_s2_’?
             std::cout << "accelerometer_m_s2: " << st.accelerometer_m_s2().at(0);
                                                       ^~~~~~~~~~~~~~~~~~
                                                       accelerometer_m_s2_
sensor_combined_Subscriber.cxx:104:37: error: ‘class sensor_combined_’ has no member named ‘accelerometer_m_s2’; did you mean ‘accelerometer_m_s2_’?
             std::cout << ", " << st.accelerometer_m_s2().at(1);
                                     ^~~~~~~~~~~~~~~~~~
                                     accelerometer_m_s2_
sensor_combined_Subscriber.cxx:105:37: error: ‘class sensor_combined_’ has no member named ‘accelerometer_m_s2’; did you mean ‘accelerometer_m_s2_’?
             std::cout << ", " << st.accelerometer_m_s2().at(2) << std::endl;
                                     ^~~~~~~~~~~~~~~~~~
                                     accelerometer_m_s2_
sensor_combined_Subscriber.cxx:106:62: error: ‘class sensor_combined_’ has no member named ‘accelerometer_integral_dt’; did you mean ‘accelerometer_integral_dt_’?
             std::cout << "accelerometer_integral_dt: " << st.accelerometer_integral_dt() << std::endl;
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~
``

which when I look into the sensor_combined_.cxx shows different naming for those variables shown below

``
sensor_combined_::sensor_combined_()
{
    // m_timestamp_ com.eprosima.idl.parser.typecode.PrimitiveTypeCode@679b62af
    m_timestamp_ = 0;
    // m_gyro_rad_ com.eprosima.idl.parser.typecode.AliasTypeCode@5cdd8682
    memset(&m_gyro_rad_, 0, (3) * 4);
    // m_gyro_integral_dt_ com.eprosima.idl.parser.typecode.PrimitiveTypeCode@d6da883
    m_gyro_integral_dt_ = 0;
    // m_accelerometer_timestamp_relative_ com.eprosima.idl.parser.typecode.PrimitiveTypeCode@45afc369
    m_accelerometer_timestamp_relative_ = 0;
    // m_accelerometer_m_s2_ com.eprosima.idl.parser.typecode.AliasTypeCode@5cdd8682
    memset(&m_accelerometer_m_s2_, 0, (3) * 4);
    // m_accelerometer_integral_dt_ com.eprosima.idl.parser.typecode.PrimitiveTypeCode@799d4f69
    m_accelerometer_integral_dt_ = 0;

}
``

TSC21, 2020-04-16:
@amano001 the msg spec for sensor_combined has change and the docs for the RTPS listener didn't. That might happen ocasionally has the listener example is not actually something actively being used - it just serves as an example.

May I ask you if you are open to open a PR and fix the naming and fields on the example on docs? Thanks

amano001, 2020-04-16:
will do, I would say the examples are definitely being used.  For someone new to this, its hard to know if I screwed up something or if the documentation has a bug (usually I think its me messing it up).