PX4/PX4-Devguide #1032: Key change need to be updated

hslyu, 2020-04-28:
According to this conversation: https://answers.ros.org/question/325039/apt-update-fails-cannot-install-pkgs-key-not-working/

The repository key has been changed as: 
old key: 421C365BD9FF1F717815A3895523BAEEB01FA116
new key: C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654

But, this is not applied to docs: https://dev.px4.io/v1.9.0/en/setup/dev_env_linux_ubuntu.html#rosgazebo

I'v tried to fork and request pull, but I can't find the markdown script corresponding to the issue.

hamishwillee, 2020-04-29:
Generally we do not update the old versions - just the master version - do you really need to target v1.9.0?

You can edit the doc for this branch: https://github.com/PX4/Devguide/blob/v1.9.0/en/setup/dev_env_linux_ubuntu.md

But I don't know what script you are referring to that is missing.

hslyu, 2020-04-29:
Thank you, close this issue

closed 2020-04-29 by hslyu