PX4/PX4-Devguide #1035: Multi-Vehicle Sim with JMAVSim: First Vehicle not linking

hamishwillee, 2020-04-30:
Ran instructions in [Multi-Vehicle Sim with JMAVSim · PX4 Developer Guide](https://dev.px4.io/master/en/simulation/multi_vehicle_jmavsim.html) for current master PX4 on Ubuntu 18.04 (in VM). 

Created 4 vehicles. All appeared to spawn but the first one did not connect to QGC - vehicles 2, 3, 4 did.
I.e. one created like `./Tools/jmavsim_run.sh -l` did not connect, but the ones created like `./Tools/jmavsim_run.sh -p 4561 -l` did.

@julianoes @Jaeyoung-Lim Any thoughts on what might have changed/can you duplicate?

Jaeyoung-Lim, 2020-04-30:
@hamishwillee I tried to reproduce this, but couldn't. - It works on my laptop

Could it be possible that something like the TCP port `4560` us already being used from a different process? Have you tried maybe rebooting and running the same commands?

hamishwillee, 2020-05-01:
Thanks @Jaeyoung-Lim . YOu were right. Reboot and rebuild worked fine.

closed 2020-05-01 by hamishwillee