PX4/PX4-Devguide #1048: ubuntu_sim_common_deps.sh downloads html instead of .tar.gz when installing FastRTPS 1.7.1 and FastCDR-1.0.8

iwishiwasaneagle, 2020-06-23:
[https://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-7-1/eprosima_fastrtps-1-7-1-linux-tar-gz](https://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-7-1/eprosima_fastrtps-1-7-1-linux-tar-gz)

[Seems like eprimosa.com changed it's links, rather than a new addition to the codebase](
https://github.com/PX4/Devguide/blame/master/build_scripts/ubuntu_sim_common_deps.sh#L43)

```
➜  ~ wget https://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-7-1/eprosima_fastrtps-1-7-1-linux-tar-gz -O eprosima_fastrtps-1-7-1-linux.tar.gz
--2020-06-23 15:32:12--  https://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-7-1/eprosima_fastrtps-1-7-1-linux-tar-gz
Resolving www.eprosima.com (www.eprosima.com)... 80.231.171.97
Connecting to www.eprosima.com (www.eprosima.com)|80.231.171.97|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘eprosima_fastrtps-1-7-1-linux.tar.gz’

eprosima_fastrtps-1-7-1-linux.tar.gz                      [  <=>                                                                                                                  ] 123.48K   432KB/s    in 0.3s    

2020-06-23 15:32:13 (432 KB/s) - ‘eprosima_fastrtps-1-7-1-linux.tar.gz’ saved [126444]

➜  ~ file eprosima_fastrtps-1-7-1-linux.tar.gz 
eprosima_fastrtps-1-7-1-linux.tar.gz: HTML document, UTF-8 Unicode text, with very long lines, with CRLF, LF line terminators
```

TSC21, 2020-06-23:
Please do not use that script to install FastRTPS and its dependencies. That part of the script should be removed. For instructions on how to install it please use https://dev.px4.io/master/en/setup/fast-rtps-installation.html.

TSC21, 2020-06-23:
Note that for binaries installation, that still apllies, since now it's called eProsima FastDDS, insteaf of FastRTPS.

iwishiwasaneagle, 2020-06-23:
Hi, can this be removed from the script then?

TSC21, 2020-06-23:
> Hi, can this be removed from the script then?

I vote yes. Please create a PR if you may. Thanks

closed 2020-06-23 by iwishiwasaneagle