PX4/PX4-Devguide #1061: RTPS/ROS2 Interface page typo

nichwall, 2020-07-23:
In the "Creating a ROS2 Advertiser" section (https://dev.px4.io/master/en/middleware/micrortps.html#creating-a-ros2-advertiser), the example file `debug_vect_advertiser.cpp` is said to be in `px4_ros_com/src/listeners` instead of `px4_ros_com/src/advertisers`. It's simple to find where the source file is actually stored, but inconsistent with documentation.

RicardoM17, 2020-10-12:
Thanks for the heads up @nichwall . I've pushed a fix in #1096 .

FYI next time if you want you can make the changes yourself. Just to make that clear, but raising awareness is the next best thing.

@hamishwillee FYI once the PR is merged, this issue can be closed.

hamishwillee, 2020-10-13:
Closed by https://github.com/PX4/Devguide/commit/6009b061dac10141ff4b799e18d16b9c4d031eb7

hamishwillee, 2020-10-13:
Thanks @RicardoM17 - much appreciated. 

@nichwall Thanks for posting this! As Ricardo says, easy to fix these yourself - if you click this icon on the top of every page it takes you to github and opens the page for editing. You can then create a PR which includes the actual change for us to review.

[image]

closed 2020-10-13 by hamishwillee