PX4/PX4-Devguide #1080: Doc Bug: Ubuntu/Debian Linux · PX4 Developer Guide

bigbellmercy, 2020-09-10:
Thanks for the [ROS/Gazebo](https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html#rosgazebo) installation method article.

When I tried it, it failed. After I changed `catkin build` to `catkin_make_isolated` in the `ubuntu_sim_ros_melodic.sh` script, it succeeded. I think the reason is that other ROS packages already installed by user in `catkin_ws/src` folder cannot be built by `catkin build` command.

According to [this](https://answers.ros.org/question/320613/catkin_make-vs-catkin_make_isolated-which-is-preferred/?answer=320706#post-id-320706), `catkin_make_isolated` is better than `catkin_make` for some cases. I think the ROS/Gazebo installation method seems to assume that ROS is not installed before in the system.

So, how about to change `catkin build` to `catkin_make_isolated`? 

Bug Page: [Ubuntu/Debian Linux · PX4 Developer Guide](https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html#rosgazebo)

TSC21, 2020-09-10:
What exactly failed?

bigbellmercy, 2020-09-10:
I forgot the error message ; ) But from my memory it seems to be the following:

>   This workspace contains non-catkin packages in it, and catkin cannot build a non-homogeneous workspace without isolation.  Try the 'catkin_make_isolated' command instead.

TSC21, 2020-09-10:
> I forgot the error message ; ) But from my memory it seems to be the following:
> 
> > This workspace contains non-catkin packages in it, and catkin cannot build a non-homogeneous workspace without isolation.  Try the 'catkin_make_isolated' command instead.

And what are those packages?

TSC21, 2020-09-10:
Also `catkin build` CAN build non-catkin packages in a workspace.

bigbellmercy, 2020-09-10:
The [ROS/Gajebo](https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html#rosgazebo) installation method worked well after I delete all the `devel_isolated`, `build_isoltated` and `install_isolated` folders in `catkin_ws` folder. But, I don't know what was the exact reason of my problem.

Sorry and thanks! I'd like to buy a coffee to you for concern.

RicardoM17, 2020-11-02:
@bigbellmercy do you feel like the original instructions were incorrect or missing something?

If not, considering that it seems to be fixed now I would close this issue. :)

hamishwillee, 2020-11-04:
Thanks for chasing this @RicardoM17 . Closing at your post got a thumbs up :-)

closed 2020-11-04 by hamishwillee

RicardoM17, 2020-11-05:
No worries @hamishwillee  and thank you also for chasing this as well. I did not receive the notification for the thumbs up.