PX4/PX4-Devguide #1086: PX4/JMAVSIM Build Error (return value 512)

KrystekT, 2020-09-22:
Hi,
I am trying to run JMAVSIM on Windows 10. It worked for me before and then something happened when I tried to run it again. I uninstalled and reinstalled the PX4 Toolchain and have tried to run it again and keep getting this error. I have look at some stuff that it has something to do with bash, dash, but I don't understand the lingo I'm pretty new to this. Here is the run error:

$ make px4_sitl jmavsim
[0/1] cd /cygdrive/c/PX4/home/Firmware/build/px4_sitl_defa...mware /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/bin/px4.exe
debugger: none
program: jmavsim
model: none
world: none
src_path: /cygdrive/c/PX4/home/Firmware
build_path: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/bin/px4.exe" "/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/cygdrive/c/PX4/home/Firmware"/test_data
INFO  [px4] Creating symlink /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/etc -> /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/bin/px4-alias.sh: line 42: `source': is a special builtin
ERROR [px4] Startup script returned with return value: 512

hamishwillee, 2020-09-23:
@KrystekT Generally best to raise first in PX4 slack #issues channel- more eyes there. They will then tell you if there is a bug and point you to right place to post. This is a docs repo, and what you have here is some problem with code, not the docs - so I am closing.

That said, @MaEtUgR FYI - in case you have seen this before.

closed 2020-09-23 by hamishwillee

MaEtUgR, 2020-10-13:
This is a known issue that came in with the NuttX upgrade and has to be solved soon but is not yet. See https://github.com/PX4/Firmware/issues/15795, https://github.com/PX4/Firmware/pull/15139#discussion_r486950808