PX4/PX4-Devguide #1104: PX4 doesn't start completely

kadhirumasankar, 2020-10-31:
Hi. Thanks for this helpful guide. I am trying to set up PX4 to work with Gazebo. I followed the instructions at https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html and then https://dev.px4.io/master/en/setup/building_px4.html.

I ran `make px4_sitl gazebo` and a Gazebo window was started with a drone in it. However, I don't think PX4 has started completely, because when I try typing something like `commander takeoff`, it just shows up as normal text, and not as input to PX4 (as in it doesn't appear like `px4> commander takeoff`.

How can I fix this? Was there some mistake with how I installed everything? Thanks in advance.

Here is the full output from my terminal:
```bash
/Firmware$ make px4_sitl gazebo
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/kadhir/Firmware/build/px4_sitl_def...ware /home/kadhir/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/kadhir/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/kadhir/Firmware
build_path: /home/kadhir/Firmware/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/home/kadhir/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/kadhir/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /opt/ros/melodic/lib:/usr/local/cuda-10.2/lib64::/home/kadhir/Firmware/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
SITL COMMAND: "/home/kadhir/Firmware/build/px4_sitl_default/bin/px4" "/home/kadhir/Firmware/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/kadhir/Firmware"/test_data
INFO  [px4] Creating symlink /home/kadhir/Firmware/build/px4_sitl_default/etc -> /home/kadhir/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 4
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  EKF2_MULTI_IMU: curr: 0 -> new: 3
  SENS_IMU_MODE: curr: 1 -> new: 0
  EKF2_MULTI_MAG: curr: 0 -> new: 2
  SENS_MAG_MODE: curr: 1 -> new: 0
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 1 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
  IMU_INTEG_RATE: curr: 200 -> new: 250
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 10.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  GPS_UBX_DYNMODEL: curr: 7 -> new: 6
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Gazebo multi-robot simulator, version 9.15.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.241
[Wrn] [Publisher.cc:185] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.
```

RicardoM17, 2020-11-02:
Hi @kadhirumasankar 

It seems that your simulation did not start (you should see gazebo poping up and your drone in view). After that is when you can `commander takeoff`

```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.241
```

What OS are you using?

I would focus on debugging this part.

kadhirumasankar, 2020-11-02:
Hi @RicardoM17 

Thanks for the reply. Gazebo starts up and the drone is in view. However, I am still not able to use `commander takeoff`

I am Ubuntu 18.04 on an Xavier. What exactly is going on in the snippet, and how would I debug that?

hamishwillee, 2020-11-03:
@kadhirumasankar It doesn't look like you're doing anything wrong.

A few confirmations 
- is this PX4 master branch? 
- Is this on a real computer or a virtual machine?
- Did you install using ubuntu.sh to install this?

If it were me, I would re-clone https://github.com/PX4/PX4-Autopilot  (note the new name) and try start it headless
```
git clone https://github.com/PX4/PX4-Autopilot --recursive
HEADLESS=1 make px4_sitl gazebo
```
If that doesn't work let me know and I'll try it myself locally. It IS possible that something broke  in master.

kadhirumasankar, 2020-11-19:
@hamishwillee I deleted the Firmware folder and did `git clone https://github.com/PX4/PX4-Autopilot --recursive`. When I run `make px4_sitl gazebo` the same thing happens as before: the simulator opens up but doesn't accept any PX4 commands. When I run `HEADLESS=1 make px4_sitl gazebo` it gets stuck on `INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560`

kadhirumasankar, 2020-11-19:
I deleted catkin_ws and the PX4 folder and tried going through installation again without activating the venv this time. Here are all the steps I followed:
Cloned `https://github.com/PX4/PX4-Autopilot`
Sourced `./Tools/setup/ubuntu.sh`
Sourced `ubuntu_sim_ros_melodic.sh`
Ran `make px4_sitl gazebo`
That gave me some errors because of pip packages that weren't installed so I installed those
But it still stops at the same point

hamishwillee, 2020-11-19:
The "NVIDIA Jetson AGX Xavier" is not a common platform; @TSC21 any advice on this one?

kadhirumasankar, 2020-12-22:
The issue was due to the firmware using a TCP port instead of a UDP port. I was able to fix it by following @luisp23's instructions at https://discuss.px4.io/t/nvidia-jetson-xavier-agx-controlling-px4-pixhawk-ros-mavros/16135/17

closed 2020-12-22 by kadhirumasankar