PX4/PX4-Devguide #1120: running offb_node.cpp

Iko-Git, 2020-11-06:
Hello, 
Could you please guide me how to run offb_node.cpp
.....
Please take it into consideration that I installed PX4 using following the below instructions:
Download PX4 Source Code:
-git clone https://github.com/PX4/PX4-Autopilot.git --recursive
Run the ubuntu.sh with no arguments (in a bash shell) to install everything:
-bash ./Tools/setup/ubuntu.sh

RicardoM17, 2020-11-06:
Hi @Iko-Git have you looked into https://dev.px4.io/master/en/ros/mavros_offboard.html ?

In a nutshell you need to add the file to your ROS package, build it, then launch a gazebo simulation, launch mavros via roslaunch (check the page after the one I sent) and then run the offbnode via rosrun.

hamishwillee, 2020-11-10:
@Iko-Git Best place to ask support questions is https://discuss.px4.io/ . This area is where you post questions about specific issues with the documentation. I am therefore closing.

closed 2020-11-10 by hamishwillee

RicardoM17, 2020-11-13:
Glad to hear it @Iko-Git 

Best of luck and have fun developing :)