PX4/PX4-Devguide #1144: ROS Script ubuntu_sim_ros_melodic.sh doesn't compatible to ubuntu 20.04

BazookaJoe1900, 2021-01-03:
ROS version Melodic is for ubuntu 18.04 only, and not for 20.04. for 20.04 we need Noetic version
but the devguide says it does fit. and might fail developers that tries to install that script: ubuntu_sim_ros_melodic.sh.
either fixing the script, or fixing the note is possible.

hamishwillee, 2021-01-14:
@BazookaJoe1900 Can you point to the problem page. 

ubuntu_sim_ros_melodic.sh is indeed not compatible with ubuntu 20.04, but my understanding is that this problem is clearly noted with the script.

BazookaJoe1900, 2021-01-14:
@hamishwillee here:
https://github.com/PX4/PX4-Devguide/blob/457200b4e27800dba8f89de4fbcc582be5ba6e7f/en/setup/dev_env_linux_ubuntu.md
`The supported/tested Linux OS versions for PX4 development are Ubuntu Linux LTS 18.04 (Bionic Beaver) and 20.04 (Focal Fossa). `

hamishwillee, 2021-01-20:
Thanks @BazookaJoe1900  - fixed in the user guide.

closed 2021-01-20 by hamishwillee