PX4/PX4-Devguide #1146: Add a new airframe: Servos positioning

JAPolo2, 2021-01-13:
I made a new airframe configuration based on the Y tricopter. The new configuration is compared to the Y tricopter in the image below. Main differences regarding the conventional Y tricopter are:

- There’s one front motor instead of two motors.
- The front motor is responsible of the 80% thrust
- The rear motors are at 90° of each other
- There are two servos to control yaw (one pear rear motor)

I created the respectif geometry (.toml), config file and mixer file as it is specified on this [post](https://github.com/PX4/PX4-Devguide/issues/435). The airframe works on the drone, however, after modifying the PID parameters several times, we still having problems with the yaw control which often oscillates. We start to think that it could be a firmware problem. To be more precise, we believe this happens because there’s no definition of the servos position in the geometry of the drone. If we’re right, I would like to know how we can specify the servos position in the firmware.
Any help will be appreciated.
Regards,

Jorge

[image]

**Geometry file**

[image]

**Mixer file**

[image]

jlecoeur, 2021-01-13:
Hi @JAPolo2, 
This is a tricky system to model with two separate mixers. The big front rotor will have a huge impact on yaw torque when actuated for pitch control. A single mixer that understands the dynamics of your robot would be needed.
I suggest you either derive the desired mixer and express it using simple mixers only, or implement a new mixer class. To simplify, I think you want to control yaw with servos only, roll with rear motors only, and pitch with rear motors only.
In the future, the control allocation module will be able to handle this platform with limited added code (see the tiltrotor example).

JAPolo2, 2021-01-13:
Hi @jlecoeur,
First of all, thanks for your answer. Yes, you're right, I want to be able to control yaw with servos only, roll with rear motors only, but pitch should be controlled by all rotors (not only by rear motors). When you say "derive the desired mixer and express it using simple mixers only", do you mean I should create a mixer file that contains only summing mixers with no geometry or that I should use a .mix for a multirotor mixer and a .aux.mix for the servos as it's defined for the FireFly6 and the CruiseAder Claire tiltrotors? Also, could you please explain how I could implement a new mixer class?

jlecoeur, 2021-01-14:
>   but pitch should be controlled by all rotors (not only by rear motors).

I suggested that as a simplification because rear rotors pitching action will have lower yaw coupling effect than the front one. Hopefully low enough that it can be countered by the servos in the feedback loop. With a more correct model, the servos are used for pitch control to prevent coupling.

>   do you mean I should create a mixer file that contains only summing mixers with no geometry or that I should use a .mix for a multirotor mixer and a .aux.mix for the servos as it's defined for the FireFly6 and the CruiseAder Claire tiltrotors? 

Summing mixer. One or two .mix files do not matter.

> Also, could you please explain how I could implement a new mixer class?

This is a bit more advanced and I think worth it only if you cannot implement your mixer with summing mixer (for example if you model the non linearity of tilting propeller). I suggest to look at the existing mixer classes in src/lib/mixers and use them as implementation reference.

hamishwillee, 2021-01-20:
Looks like this has been resolved, so I am closing. If you wish to continue discussion please post in https://discuss.px4.io/.

closed 2021-01-20 by hamishwillee