PX4/PX4-Devguide #1148: Collision prevention not working on PX4 V5nano board

AleksejSen, 2021-01-28:
SW version: 1.11
I followed the guide : https://docs.px4.io/master/en/computer_vision/collision_prevention.html
and tried with Tfmini(changed in the drivers to face forward and recompiled) and CM8jl65 facing forward(changed in the driver), I see the distance reading in QGC Mavlink inspector under under DISTANCE_SENSOR but the drone continues to fly forward ignoring CP_DIST (in POSITIONING mode controlled by the joystick)
Current parameters settings:
CP_DELAY: 0.400
CP_DIST: 2.0 M (tried 3m also)
CP_GO_NO_DATA: tested bout 1 and 0
CP_GUIDE_ANG 30.deg
SENS_CM8JL65_R0: ROTATION_FORWARD_FACING

dagar, 2021-01-28:
Can you share the flight log? https://logs.px4.io/

dagar, 2021-01-28:
@jkflying

jkflying, 2021-01-28:
As @dagar says, need a log to help you, and you need to turn on the computer vision topics in SDLOG_PROFILE . Also, just a tip, for a directly connected distance sensor you can set CP_DELAY to 0

AleksejSen, 2021-01-28:
[flight_log.tar.gz](https://github.com/PX4/PX4-Devguide/files/5888328/flight_log.tar.gz)

AleksejSen, 2021-01-28:
will turn it on next time i'll fly then

hamishwillee, 2021-02-03:
FYI, I'm slowly trying to kill this repo. Issues should best go to https://github.com/PX4/PX4-user_guide/issues