PX4/PX4-Devguide #1149: Damaged power port on Pixhawk 1

MJ1096p, 2021-03-16:
Hi Team,

Hope your doing great. 

Im in need of some help regarding the above the subject. The Power port on my  Pixhawk 1 Flight Controller isnt working.

When power module is plugged into the power port pixhawk doesn't turn on.
( * power module works fine. I checked with a multimeter and also tried plugging in power module to other ports like Serial, Telemetry and GPS ; Pixhawk turns On.    But obviously that's not a permanent solution
* I plugged in the usb cable and pixhawk powered On.    Also not a permanent solution)
Hence from above points its clear that problem exists only in the power port and not the entire pixhawk 1 flight controller.

Me being a layman in this field I request you to help by throwing some light on the issue and Stating which Components need to be Replaced so as to power the pixhawk via power module and use its function. Looking forward to hearing from you

Best Regards
M J

davids5, 2021-03-16:
@MJ1096p  this is not a dev guide issue, it is a pixhawk and would be better suted posted in https://github.com/pixhawk/Hardware

Anything on the yellow line, or connected to is is suspect.

[image]

MJ1096p, 2021-03-16:
Hi @davids5 ,
Thanks a lot for the response with schematic and pcb layout.
But I quite dint understand  :(
Also i have the PX4FMUV2.4.8 ( pixhawk 2.4.8 ) .
Can you please have a look at the pictures I have uploaded and respond correspondingly.
Looking forward to hearing from you
Best Regards
[image]
[image]

davids5, 2021-03-16:
@MJ1096p - The looks like a clone, without any of the HW it is supposed to have for the power supply.  Use an ohm meter, trace from the power connector it to all diodes, and Resistors looking for opens. Other then that, I have nothing else to offer.

hamishwillee, 2021-03-17:
Closing this issue. @MJ1096p If you need to discuss further then best place is discussion boards or https://github.com/pixhawk/Hardware as suggested by David.